bm16ton/ft2232-mpsse-i2c-spi-kern-drivers #1: how you drive FBTFT LCD display using the FT232H chip?

gadh, 2022-06-20:
the LCD display should be driven by SPI commands. and you use the FT232 USB to SPI chip. how you translate the SPI commands to USB  commands in order to control the LCD ? i'de be happy if you can elaborate on this. i want to do the same in the most straight forward and simple way i can. if i can just send an image to the USB port and it will be displayed on the LCD it would be best.

bm16ton, 2022-06-20:
i took the fbtft folder from an older kernel source tgat still had
fbtft_device. Fbtft_device was a kernel driver that would help set the
platform data of a lcd tft device and load the appropriate driver etc. I
couldnt get the platform data to pass without it. After using/incorperating
the older fbtft stack into my kernel, you just need to edit the
ft232h-intf.c, i believe i have everything needed for code already in tge
source just may need to uncomment it. The other option is the drm driver
for the ili9341 which takes the platform data perfectly. Tho the drm driver
worked great it only worked on a specific kernel version and ibdont
remember which one then it stopped. Never had time to look into why it
stopped working i kinda remember a second driver being introduced that may
have been causing a conflict. im thinking one of kernel sources has the
older fbtft stack alrwady integrated. If it would help when i get a sec
(possibly later today) i could rig uo a kernel source with everything for
an ili9311 and fbtft and post it for yeah.

gadh, 2022-06-20:
thanks a lot for your quick response !
i work with kernel 4.18 if it helps, can work also with 5.18 if its needed.
can you explain how the LCD driver works with the FT232H ? did you convert the SPI commands to USB commands for it to work ?

bm16ton, 2022-06-20:
ah i have a driver (well two actually ill explain)  i have aa kernel driver
for the ft232h/ft2232h/ft4232  for the 2232/4232 series when plugged in the
forst interface gets the spi driver which allows the ftdinto be a general
purpose spi host just like on say a raspberry pi so, only difference is
instead of using devicetree entries everything is done in driver do need to
edit the driver to tell it u have a lcd attached and then use the ili9341
kernel driver from fbtft or tge new drm driver..or any other spi device
that has a kernel driver. the second interface is i2c driver that can be
used with any i2c device with a kernel driver. the spi driver alsobis a
gpio driver and i recently added irq support (via polling) so any spi
device that needs extra gpio/gpio with irq, support now works. This is all
done by setting the eeprom string to ft2232h-16ton or ft4232h-16ton and
that tells the linux ftdi serial driver to ignore them (except the 3rd 4th
serial interfaces on the 4232 those serial ports still work) since tge
ft232h only had one interface i was using the eeprom string to switch
between i2c and spi which became annoying so i chnaged it for the 232h, now
when plugged in a simple gui comes up asking if u want it to be a usb to
serial or spi or i2c device plus sum simple settings for each option
whichni found much easier. Which device are you using? I didnt even check
where these emails started from i assumed my kernel driver source for ftdi
but if nnot ill send u a link, but since the ftdi serial driver woukd need
patching as well it makes out of tree compilingless effective (you could
just blacklist ftdi-sio but hell lets do it right! lol) i havent tried
compiling my drivers for a kernel as old as 4.18 i think the earliest was
maybe 5.9ish and works all the way up to 5.18 (altho now thinking on it
tgat was before the irq additions) i currently use 5.12 on daily driver
(arm64 laptop all patched out and doesnt work great on newer kernels) and
5.18 on a rarely used x86. Let me know if u prefer 5.12 or 5.18 and as soon
as i can (hopefully tonight) ill setup my rig and get the kernel driver
setup for the lcd. i only have an ili9341 so fingers crossed its what your
using. i could get tge kernel driver side down for any lcd supported by
fbtft but the module parameters for fbtft_device on another lcd would be
hardware dependant. Hurumph sorry for the long email and broken touchscreen
typing but uopefully that helped..Let me know glad to help!

gadh, 2022-06-21:
thanks alot for your help !
i prefer kernel 5.18 and you can use your screen , i'll make the needed adjustments for my 9163 LCD.
i prefer not to use the pop up GUI, just use plain SPI from the beginining.
waiting for you !