bm16ton/ft2232-mpsse-i2c-spi-kern-drivers #2: Binding I2C Driver to USB Device

ve-nt, 2022-08-01:
Hi, I have a PCB with an FT2232HQ connected to an embedded Linux host over USB. I want to use the FT2232HQ's I2C interface. I have patched your driver onto my kernel, and can see the driver present in `/sys/bus/usb/drivers/i2c-ftdi`, and when I plug in my FT2232HQ I can see it enumerate as a USB device, however the driver does not bind to the device. I have also tried setting the `new_id` file to the VendorID and ProductID of the FT2232HQ, but that hasn't binded the driver either. How did you get your driver to bind to your device?

bm16ton, 2022-08-02:
ah did u program the eeprom with the 16ton name? does it show up in
i2cdetect -l? if its loading and showing in i2cdetect then modinfo the
kernel kodule u want to bind and pick an alias name from the list, so the
command "modinfo modulename" then just use sysfs to bind it so a tmp102
driver at address 0x48 and i2cdetect showing ftdi on i2c-7 it woukd be
"echo tmp102 0x48 | sudo tee /sys/class/i2c-adapter/i2c-7/new_device"

ve-nt, 2022-08-02:
Okay, after modifying the code so that it also looks for "Dual RS232-HS" and treats it the same as the 2232-I2C device, I have the driver successfully binding! This is only a temporary workaround, changing the actual product name would be better I think so I could still use some pointers on that.

bm16ton, 2022-08-14:
crap didnt see this till now, i sent an email once explaining let me see if i can find it

bm16ton, 2022-08-14:
the email contents
Ok I attached some of my conf files for 232h 2232h 4232h, also an example conf I found online some time ago. Each series of chip which includes all the regular usb to serial have own options and differ from the others. A complete list or even partiallist I havent found so I added a few comments. Bigges thing I wouldnt change vendor_id or product_id for multiple reasons includin my drivers and practically everything else will no longer work. You can specify the type of eeprom but unless youre certain of the model I would leave it commented out. The only changes needed for my drivers are the product name near the top of the conf file, and the only other general change is making sure you have a filename listed on the last line, whatever filename you pick will be what the program uses to save and/or write the firmware from. The editin of eeprom is the only way to enabl certain modes IE ft245/high speed opto and a couple others, also can set pin drive strength, make certain pins control rx/tx led control pins,or put non mpsse pins into gpio mode on ft232h etc. Ive found commit logs and the code itself the best place to try and find the various options for the various boards but its tedious and confusing, hopefully my examples help. So after installing libftdi and ftdi_eeprom (I  believe part of source package for libftdi or seperate install via package managers). We go to whatever directory you have your editd conf file in terminal.  and we run "ftdi_eeprom --verbose --build-eeprom ft232h.conf" if u renamed ft232h.conf make sure you use the new name. What ever filename you picked for the last line in conf the app should have created with all the settings from the config IE mine is ft232h.bin so that file should now exist. Now run the app again "ftdi_eeprom --verbose --flash-eeprom ft232h.conf" It will now flash the eeprom, and I would run it again on the first flashing. Now it is required to unplug and replug the ftdi device for complete re-enumeration. To verify it took obviously dmesg and lsusb wll show different  but its good idea to run "ftdi_eeprom --verbose --read-eeprom ft232h.conf" to see the current settings in full. A word of caution, several apps exist for eeprom burning the ftdi chips but it is possible if something goes wrong that it will leave the device in a inoperable state until the eeprom is physically removed or prgrammed via external programmer somehow with ftdi chip in reset. Ive made plenty of mistakes and never had that happen but I would only use the official app or lbftdi's they seem rock solid. I feel like probly more tidbits that ould be helpful but cant think of any at the moment. People seem to think setting if the board is bus powered or self powered is important but it never mattered to me. If your ft232h is a adafruit board I put 2 lines currently commented out that make sure to set the pin they use for the power led to on, they have a second led ttached to another pin that can also be used as general purpose IE rx tx indicators (actually nothing about the power le is power related its jst a generic pin so can be repurposed) I think thats it, also you dont need to edit m configs at all if you dont want to, isnt anything board specific as I recall in the ft232h.conf (or probly the others either). Crap one last thing Ive had multiple copies of ftdi_eprom installed at same time so ftdi_eeprom may not be the executables name tho it will be very similiar so dont let that throw yeah. Let me know if you have any issues or questions! Hope it all works out!

and basic config contents for ft2232h
vendor_id=0x0403    # Vendor ID
product_id=0x6010
#default_pid=0x6010  # used to change/revert pid/vid values
max_power=0       # Max. power consumption: value * 2 mA. Use 0 if self_powered = true.

###########
# Strings #
###########
manufacturer="FTDI"         # Manufacturer
product="ft2232H-16ton"  # Product
serial="16ton"             # Serial

###########
# Options #
###########
self_powered=true  # Turn this off for bus powered
remote_wakeup=false # Turn this on for remote wakeup feature
use_serial=true     # Use the serial number string

# Normally out don't have to change one of these flags
in_is_isochronous=false     # In Endpoint is Isochronous
out_is_isochronous=false    # Out Endpoint is Isochronous
suspend_pull_downs=false    # Enable suspend pull downs for lower power
change_usb_version=false    # Change USB Version
usb_version=0x0200          # Only used when change_usb_version is enabled

# _type options are UART, FIFO, OPTO, CPU, FT1284
#cha_vcp=false
#cha_type=FIFO
#cha_vcp=true
#cha_type=UART
#chb_vcp=true
#chb_type=FIFO
#cha_rs485=true

########
# Misc #
########

filename="ft2232h.bin"   # Filename, leave empty to skip file writing