bm16ton/ft2232-mpsse-i2c-spi-kern-drivers #4: I2C with C232HM-DDHSL-0 - success with hack

mungewell, 2023-10-26:
I have a cable which is labelled 'C232HM-DDHSL-0', by default this loads the stock 'ftdi-sio' driver and would NOT load your I2C driver.

It seems that it has a different product ID
```
$ lsusb -v | grep -B 5 -A 5 FTDI
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0403 Future Technology Devices International, Ltd
  idProduct          0x6014 FT232H Single HS USB-UART/FIFO IC
  bcdDevice            9.00
  iManufacturer           1 FTDI
  iProduct                2 C232HM-DDHSL-0
  iSerial                 3 FT7IT1Y5
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
```

Patching the i2c driver with this, and blacklisting the 'ftdi_sio' driver, enabled it to be found when plugged and I was able to scan my target devices.
```
diff --git a/i2c-ftdi/i2c-ftdi.c b/i2c-ftdi/i2c-ftdi.c
index ff29fe9..2e871bf 100755
--- a/i2c-ftdi/i2c-ftdi.c
+++ b/i2c-ftdi/i2c-ftdi.c
@@ -876,6 +876,12 @@ static int ftdi_usb_probe(struct usb_interface *interface,
                        ftdi_usb_delete(ftdi);
                                return -ENODEV;
                }
+       } else if (ftdi->udev->product && !strcmp(ftdi->udev->product, "C232HM-DDHSL-0")) {
+               ret = ftx232h_single_probe(interface);
+               if (ret < 0) {
+                       ftdi_usb_delete(ftdi);
+                               return -ENODEV;
+               }
        } else if (ftdi->udev->product && !strcmp(ftdi->udev->product, "ft232H-16ton-spi")) {
         dev_info(&interface->dev, "Ignoring single spi  Interface\n");
                ftdi_usb_delete(ftdi);
```

```
root@the-void:/sys/bus/i2c/devices/i2c-21# ls -al
total 0
drwxr-xr-x 4 root root    0 Oct 25 20:50 .
drwxr-xr-x 7 root root    0 Oct 25 20:50 ..
--w------- 1 root root 4096 Oct 25 20:50 delete_device
lrwxrwxrwx 1 root root    0 Oct 25 20:50 device -> ../../1-1:1.0
drwxr-xr-x 3 root root    0 Oct 25 20:50 i2c-dev
-r--r--r-- 1 root root 4096 Oct 25 20:50 name
--w------- 1 root root 4096 Oct 25 20:50 new_device
drwxr-xr-x 2 root root    0 Oct 25 20:50 power
lrwxrwxrwx 1 root root    0 Oct 25 20:50 subsystem -> ../../../../../../../bus/i2c
-rw-r--r-- 1 root root 4096 Oct 25 20:50 uevent
root@the-void:/sys/bus/i2c/devices/i2c-21# i2cdetect 21
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-21.
I will probe address range 0x08-0x77.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- 24 -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- 56 57 -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- 73 -- -- -- --                         
root@the-void:/sys/bus/i2c/devices/i2c-21# cat name
FTDI USB-to-I2C at bus 001 device 014
```

Thanks :-)

bm16ton, 2023-10-26:
Awesome! I have been sticking with using custom values/names so that it
wouldn't interfere with peoples normal operation but I can def add
something for this chip. Is it a usb-hs device? How many gpio? Reminds me I
gotta add mpsse gpio to the i2c driver someday. Thanks again that's awesome!

mungewell, 2023-10-26:
The 'device' is a debug cable from FTDI themselves, I got it from Digikey.
https://www.digikey.ca/en/products/detail/ftdi-future-technology-devices-international-ltd/C232HM-DDHSL-0/2714139

There's also a 5v version.
https://www.digikey.ca/en/products/detail/ftdi-future-technology-devices-international-ltd/C232HM-EDHSL-0/2714140

My main reason to report was mostly to leave a bread-crumb trail for my future self should I forget and have to repeat... but since these are 'official' cables perhaps that is reason enough to add the IDs.

bm16ton, 2023-10-27:
Very cool, I'm guessing they are using ft232h. Yeah I didn't want any
contention between ftdi-sio and my drivers is why I used the usb descriptor
hack by patching the ftdi-sio driver to ignore the 16ton product id's, the
spi/i2c drivers use the various product ids to determine gpio count etc. Oh
crap did I leave out the ftdi-sio patch from the repo? i gotta check
that..also do a small write up about programming eeprom I get a lot of
emails about that.