bm16ton/ft2232-mpsse-i2c-spi-kern-drivers #6: FT232H - No activity on data line

strobe84, 2024-02-28:
Good afternoon. Firstly, thank you thank you thank you for developing this driver - it's very much appreciated. I have an Adafruit FT232H based USB-i2c adapter which I'm intending on using for testing some prototype PCB's ahead of them being packaged into complete units. I thought initially these boards and chip would be ideal, without realising there was no mainline kernel driver! I've programmed the EEPROM on the board with the "16ton" serial etc, the driver seems to bind to the device and is enumerated, but I cannot seem to identify anything on the bus using i2cdetect. To validate the setup I've checked what's "actually" there using PyFTDI which seems to return a number of devices, but using the Kernel driver does not. Having probed the clock and data lines, I'm seeing a solid clock with voltage swings of ~1.6V peak to peak(1.8V logic on this board) but nothing on the data line whatsoever. What am I missing? dmesg output as follows:

`[ 8025.610752] usb 1-2: new full-speed USB device number 11 using ohci-pci
[ 8025.895275] usb 1-2: config 1 interface 0 altsetting 0 endpoint 0x81 has invalid maxpacket 512, setting to 64
[ 8025.895280] usb 1-2: config 1 interface 0 altsetting 0 endpoint 0x2 has invalid maxpacket 512, setting to 64
[ 8025.923191] usb 1-2: New USB device found, idVendor=0403, idProduct=6014, bcdDevice= 9.00
[ 8025.923195] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8025.923197] usb 1-2: Product: ft232H-16ton
[ 8025.923198] usb 1-2: Manufacturer: FTDI
[ 8025.923199] usb 1-2: SerialNumber: 16ton
[ 8025.934452] i2c-ftdi 1-2:1.0: probe i2cftdi freq 100000
[ 8026.028619] i2c_dev: adapter [FTDI USB-to-I2C at bus 001 device 011] registered as minor 1
[ 8026.028646] i2c i2c-1: adapter [FTDI USB-to-I2C at bus 001 device 011] registered
[ 8026.028653] i2c-ftdi 1-2:1.0: Initialized FTDI-based I2C device
[ 8026.474508] usbcore: registered new interface driver usbserial_generic
[ 8026.474516] usbserial: USB Serial support registered for generic
[ 8026.486995] usbcore: registered new interface driver ftdi_sio
[ 8026.487006] usbserial: USB Serial support registered for FTDI USB Serial Device
[ 8027.218245] i2c i2c-1: ioctl, cmd=0x705, arg=0x7fff646f9718
[ 8027.218337] i2c i2c-1: ioctl, cmd=0x703, arg=0x08
[ 8027.218341] i2c i2c-1: ioctl, cmd=0x720, arg=0x7fff646f96a0
[ 8090.731220] i2c i2c-1: ioctl, cmd=0x705, arg=0x7ffee6739448
[ 8090.731312] i2c i2c-1: ioctl, cmd=0x703, arg=0x08
[ 8090.731317] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[ 8101.115436] i2c i2c-1: ioctl, cmd=0x703, arg=0x09
[ 8101.115443] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[ 8111.365576] i2c i2c-1: ioctl, cmd=0x703, arg=0x0a
[ 8111.365584] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[ 8121.597206] i2c i2c-1: ioctl, cmd=0x703, arg=0x0b
[ 8121.597214] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[ 8131.850037] i2c i2c-1: ioctl, cmd=0x703, arg=0x0c
[ 8131.850044] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[ 8142.099274] i2c i2c-1: ioctl, cmd=0x703, arg=0x0d
[ 8142.099282] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[ 8152.330493] i2c i2c-1: ioctl, cmd=0x703, arg=0x0e
[ 8152.330500] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[ 8162.560939] i2c i2c-1: ioctl, cmd=0x703, arg=0x0f
[ 8162.560946] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[ 8172.815488] i2c i2c-1: ioctl, cmd=0x703, arg=0x10
[ 8172.815496] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[ 8183.065416] i2c i2c-1: ioctl, cmd=0x703, arg=0x11
[ 8183.065423] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[ 8193.275958] i2c i2c-1: ioctl, cmd=0x703, arg=0x12
[ 8193.276122] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[ 8203.531151] i2c i2c-1: ioctl, cmd=0x703, arg=0x13
[ 8203.531159] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[ 8213.762652] i2c i2c-1: ioctl, cmd=0x703, arg=0x14
[ 8213.762683] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[ 8223.997347] i2c i2c-1: ioctl, cmd=0x703, arg=0x15
[ 8223.997354] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[ 8234.249359] i2c i2c-1: ioctl, cmd=0x703, arg=0x16
[ 8234.249367] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[ 8244.483014] i2c i2c-1: ioctl, cmd=0x703, arg=0x17
[ 8244.483021] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[ 8254.735057] i2c i2c-1: ioctl, cmd=0x703, arg=0x18
[ 8254.735064] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[ 8265.211592] i2c i2c-1: ioctl, cmd=0x703, arg=0x19
[ 8265.211600] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[ 8275.467230] i2c i2c-1: ioctl, cmd=0x703, arg=0x1a
[ 8275.467238] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[ 8285.703307] i2c i2c-1: ioctl, cmd=0x703, arg=0x1b
[ 8285.703314] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[ 8295.956322] i2c i2c-1: ioctl, cmd=0x703, arg=0x1c
[ 8295.956329] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[ 8306.186121] i2c i2c-1: ioctl, cmd=0x703, arg=0x1d
[ 8306.186151] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[ 8316.418025] i2c i2c-1: ioctl, cmd=0x703, arg=0x1e
[ 8316.418033] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[ 8326.670850] i2c i2c-1: ioctl, cmd=0x703, arg=0x1f
[ 8326.670860] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffee67393d0
[... 59 lines cut ...]
[ 8680.067723] i2c i2c-1: ioctl, cmd=0x703, arg=0x08
[ 8680.067727] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffe62c25520
[ 8690.442427] i2c i2c-1: ioctl, cmd=0x703, arg=0x09
[ 8690.442434] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffe62c25520
[ 8700.674432] i2c i2c-1: ioctl, cmd=0x703, arg=0x0a
[ 8700.674439] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffe62c25520
[ 8710.923602] i2c i2c-1: ioctl, cmd=0x703, arg=0x0b
[ 8710.923609] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffe62c25520
[ 8721.153758] i2c i2c-1: ioctl, cmd=0x703, arg=0x0c
[ 8721.153765] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffe62c25520
[ 8731.389268] i2c i2c-1: ioctl, cmd=0x703, arg=0x0d
[ 8731.389276] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffe62c25520
[ 8741.640175] i2c i2c-1: ioctl, cmd=0x703, arg=0x0e
[ 8741.640202] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffe62c25520
[ 8751.871220] i2c i2c-1: ioctl, cmd=0x703, arg=0x0f
[ 8751.871228] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffe62c25520
[ 8762.126635] i2c i2c-1: ioctl, cmd=0x703, arg=0x10
[ 8762.126642] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffe62c25520
[ 8772.362301] i2c i2c-1: ioctl, cmd=0x703, arg=0x11
[ 8772.362308] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffe62c25520
[ 8782.590369] i2c i2c-1: ioctl, cmd=0x703, arg=0x12
[ 8782.590377] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffe62c25520
[ 8792.842249] i2c i2c-1: ioctl, cmd=0x703, arg=0x13
[ 8792.842257] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffe62c25520
`

strobe84, 2024-02-28:
Also, just looking at the details of the loaded modules, it's showing the i2c_ftdi module as live and set the "bind232h"

> Module = "i2c_ftdi"

  Attributes:
    coresize            = "20480"
    initsize            = "0"
    initstate           = "live"
    refcnt              = "0"
    srcversion          = "7C78FE5AE080A84FE3F2B41"
    taint               = ""
    uevent              = <store method only>

  Parameters:
    100k                = "1"
    1m                  = "0"
    3m                  = "0"
    adaptive            = "0"
    bind232h            = "1"

  Sections:
    .bss                = "0xffffffffc057c4c0"
    .data               = "0xffffffffc057c000"
    .exit.text          = "0xffffffffc057a3b8"
    .gnu.linkonce.this_module= "0xffffffffc057c140"
    .init.text          = "0xffffffffc057f000"
    .note.Linux         = "0xffffffffc057b024"
    .note.gnu.build-id  = "0xffffffffc057b000"
    .rodata             = "0xffffffffc057b2a0"
    .rodata.str1.1      = "0xffffffffc057b0e4"
    .rodata.str1.8      = "0xffffffffc057b188"
    .strtab             = "0xffffffffc0580c60"
    .symtab             = "0xffffffffc0580000"
    .text               = "0xffffffffc0579000"
    .text.unlikely      = "0xffffffffc057a0b6"
    __mcount_loc        = "0xffffffffc057b054"
    __param             = "0xffffffffc057b3b8"

> Module = "ftdi_sio"

  Attributes:
    coresize            = "65536"
    initsize            = "0"
    initstate           = "live"
    refcnt              = "0"
    srcversion          = "53F5DAE66F970B47AAC3F8C"
    taint               = ""
    uevent              = <store method only>

  Parameters:
    bind232h            = "0"
    dont_ignore         = "0"
    ndi_latency_timer   = "1"

  Sections:
    .bss                = "0xffffffffc05fcfc0"
    .data               = "0xffffffffc05fc000"
    .exit.text          = "0xffffffffc05f2356"
    .gnu.linkonce.this_module= "0xffffffffc05fcc40"
    .init.text          = "0xffffffffc0600000"
    .note.Linux         = "0xffffffffc05f3024"
    .note.gnu.build-id  = "0xffffffffc05f3000"
    .retpoline_sites    = "0xffffffffc05fa838"
    .rodata             = "0xffffffffc05f39e0"
    .rodata.str1.1      = "0xffffffffc05f3768"
    .rodata.str1.8      = "0xffffffffc05f31b0"
    .strtab             = "0xffffffffc0607738"
    .symtab             = "0xffffffffc0601000"
    .text               = "0xffffffffc05ef000"
    .text.unlikely      = "0xffffffffc05f1f38"
    __dyndbg            = "0xffffffffc05fc528"
    __jump_table        = "0xffffffffc05fb000"
    __mcount_loc        = "0xffffffffc05f3054"
    __param             = "0xffffffffc05fa7c0"

bm16ton, 2024-02-29:
Hello good sir! So do i see ftdi_sio driver also being loaded? Crap android
doesnt show youremail while typing mine but your using ft232h? If so (im
sorry ive been awake for a longtime but pretty sure the ft232h only has a
single channel) that seems strange. Did you patch ftdio_sio as well, so
itll ignore the 16ton named devices. Also it dropped packet size from 512
to 64 so its not on a usb2 highspeed port, and jeez im not sure if i ever
tried these drivers on non highspeed ports before, its possible i hard
coded something highspeed specific somewhere. Id start with blacklisting
ftdi_sio. When you run i2cdetect -l  (to list i2c  master devuces) does
that show the ftdi i2c and then fail when running i2cdetect on that
particular i2c bus? Or does it fail to even see the ftdi i2c bus?
Unfortunatly i need to goto bed, but i promise you will have my complete
attention tomorrow until we get you going! Ive been wanting to get back to
these drivers with some tlc, and tomorrow sounds like the day;) I should be
up and about by 8am eastern time, and available all day. Sorry i have to
cut it short tonight, but i got yeah. Im curious if you tried tge ft232h
gui i never realy finished adding features etc, but that made it easier ti
switch the single interface between protocols. Fingers crossed its just
ftdi_sio which shouldnt be loaded, or usb-fullspeed.

strobe84, 2024-02-29:
Good morning! I've blacklisted ftdi_sio and that doesn't seem to have had an effect(other than the driver no longer loading). The net effect of all this is that i2cdetect appears to work if you point it at /dev/i2c-1, but simply times out at every device address, which makes sense as I can see nothing on the i2c data line using a probe.

I've not tried the UI yet as I'm trying to avoid the use of a GUI on this machine if I can help it, but I did install it(after messing around with the gtkdialog build for a good amount of time!).

Appreciate your help! I refuse to bow down to using circuitpython to drive these things!

strobe84, 2024-02-29:
So, for funsies I thought I'd hook the RTO up again and made an interesting discovery... When using the i2c kernel driver, I don't see anything on the data line at all.... however, I see what appears to be data on the clock line(16ton-kernel-driver-clock-output.png). As a comparison I thought I'd hook the adapter up to my VM running the standard ftdi_sio driver and libftdi, driving it via Python. The clock line now actually looks like a clock(libfdi-lib-python-clock-output.png) and there's data on the data line(libfdi-lib-python-data-output.png) so seems like there's something amiss here!
[image]
[image]
[image]

bm16ton, 2024-02-29:
Cool were making progress! Im surprised it loads as bus 1 usually more i2c
busses on a pc then that. Okay so if you are using a ft232h, it actually
has true open drain pins unlike the other models. Its a long shot but if
the userland libs you tested with put it into push/pull you could get by
without adding pullup resistors, and possibly the kernel driver is in open
drain and would need them...again very very long shot but worth eliminating
as possability. Oh and its worth asking is the data pin staying high or
low? Also do you have any other mpsse devices IE ft2232 or ft4232? just
curious for testing. Is your ft232h an adafruit or generic? Which kernel
version are you running? Haha sorry for the slew of questions

strobe84, 2024-02-29:
In answer to your questions:

> Which kernel
version are you running? 

5.18.5

> Is your ft232h an adafruit or generic?

it's one of these: https://www.adafruit.com/product/2264

> Also do you have any other mpsse devices IE ft2232 or ft4232?

Sadly not :(

> Oh and its worth asking is the data pin staying high or
low?

It stays high.

> Okay so if you are using a ft232h, it actually
has true open drain pins unlike the other models. Its a long shot but if
the userland libs you tested with put it into push/pull you could get by
without adding pullup resistors, and possibly the kernel driver is in open
drain and would need them...again very very long shot but worth eliminating
as possability.

I beleive the latest iteration doesn't require the pullups(as above) and it's now a bit of a pig to add them in all honesty!

bm16ton, 2024-02-29:
Its awesome I have the same adafruit board here! Im running the newest
kernel 6.8-rc and I programmed the eeprom product to "ft232H-16ton-i2c" but
having it "ft232H-16ton" along with the bind module parameter should equal
to the same thing. Ill try and get a kernel version 5.18.5 compiled and see
if it still works for me. This is a silly question but i make silly
mistakes all the time so ill ask (please forgive me) but are you using the
qwiic connector or the headers for the i2c? I just wanna make sure you have
the switch to turn on the i2c (well not turn on, but combine the 2 pins
used for rx and tx together), needless to say ive forgotten that more then
once. Oh wait nevermind you said it worked using the userland libs, so cant
be that. Well let me throw mine on an fx2 and ill sigrock a bus scan and at
least compare to the pics you sent me. If possible I wouldnt mind seeing
the outputs of i2cdetect -l and i2cdetect -y 1, doubtful itll show anything
but cant hurt. Ill download the kernel source now. Oh and what i2c devices
do you have attached to your board? Maybe i have the same and can get as
close to your setup as possible. Might not hurt to peak at your eeprom conf
file as well

strobe84, 2024-02-29:
> Also now, randomly, I'm not seeing the device map to /dev/i2c-1 any more but it is detected by the kernel
> 
> > [ 1108.453396] usb 1-2: new full-speed USB device number 9 using ohci-pci
> > [ 1108.764269] usb 1-2: config 1 interface 0 altsetting 0 endpoint 0x81 has invalid maxpacket 512, setting to 64
> > [ 1108.764282] usb 1-2: config 1 interface 0 altsetting 0 endpoint 0x2 has invalid maxpacket 512, setting to 64
> > [ 1108.793426] usb 1-2: New USB device found, idVendor=0403, idProduct=6014, bcdDevice= 9.00
> > [ 1108.793430] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> > [ 1108.793431] usb 1-2: Product: ft232H-16ton
> > [ 1108.793433] usb 1-2: Manufacturer: FTDI
> > [ 1108.793434] usb 1-2: SerialNumber: 16ton
> 
> > monitor will print the received events for:
> > UDEV - the event which udev sends out after rule processing
> > KERNEL - the kernel uevent
> 
> KERNEL[1165.232804] remove /devices/pci0000:00/0000:00:06.0/usb1/1-2/1-2:1.0 (usb) KERNEL[1165.233059] unbind /devices/pci0000:00/0000:00:06.0/usb1/1-2 (usb) KERNEL[1165.233127] remove /devices/pci0000:00/0000:00:06.0/usb1/1-2 (usb) UDEV [1165.234973] remove /devices/pci0000:00/0000:00:06.0/usb1/1-2/1-2:1.0 (usb) UDEV [1165.236106] unbind /devices/pci0000:00/0000:00:06.0/usb1/1-2 (usb) UDEV [1165.236772] remove /devices/pci0000:00/0000:00:06.0/usb1/1-2 (usb) KERNEL[1169.815329] add /devices/pci0000:00/0000:00:06.0/usb1/1-2 (usb) KERNEL[1169.825176] add /devices/pci0000:00/0000:00:06.0/usb1/1-2/1-2:1.0 (usb) KERNEL[1169.825344] bind /devices/pci0000:00/0000:00:06.0/usb1/1-2 (usb) UDEV [1170.234868] add /devices/pci0000:00/0000:00:06.0/usb1/1-2 (usb) UDEV [1170.248616] add /devices/pci0000:00/0000:00:06.0/usb1/1-2/1-2:1.0 (usb) UDEV [1170.250507] bind /devices/pci0000:00/0000:00:06.0/usb1/1-2 (usb)

Ignore this... for some reason the drive had loaded without the bindft232 or whatever parameter... FML

strobe84, 2024-02-29:
> This is a silly question but i make silly
mistakes all the time so ill ask (please forgive me) but are you using the qwiic connector or the headers for the i2c?

I'm just soldered direct to the thru-hole pads on the board.

> If possible I wouldnt mind seeing
the outputs of i2cdetect -l and i2cdetect -y 1, doubtful itll show anything
but cant hurt.

`testuser@PCB0006-1:~$ sudo i2cdetect -l
[sudo] password for testuser:
i2c-0   smbus           SMBus PIIX4 adapter at 4100             SMBus adapter
i2c-1   i2c             FTDI USB-to-I2C at bus 001 device 013   I2C adapter
`

> 

> `testuser@PCB0006-1:~$ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
`

bm16ton, 2024-02-29:
Do you have a usb2-highspeed port you could try? I unfortunately have
no usb2-fullspeed hubs to test with, so I have no idea if that can
cause issue or not. Kinda curious now lol

strobe84, 2024-02-29:
> [Thu Feb 29 15:13:26 2024] usb 1-1: reset high-speed USB device number 4 using ehci-pci
[Thu Feb 29 15:13:27 2024] i2c-ftdi 1-1:1.0: probe i2cftdi freq 400000
[Thu Feb 29 15:13:27 2024] i2c_dev: adapter [FTDI USB-to-I2C at bus 001 device 004] registered as minor 1
[Thu Feb 29 15:13:27 2024] i2c i2c-1: adapter [FTDI USB-to-I2C at bus 001 device 004] registered
[Thu Feb 29 15:13:27 2024] i2c-ftdi 1-1:1.0: Initialized FTDI-based I2C device

bm16ton, 2024-02-29:
I take it that nothing is different on the highspeed port? Also I should
ask did you incorporate the driver into your kernel source or build it outa
tree? And Its been a while since Ive updated the drivers so the source is
most likely current. I have 5.18 compiling now, fingers crossed. What i2c
devices do you have attached to the ft232h?

strobe84, 2024-02-29:
> I take it that nothing is different on the highspeed port? Also I should ask did you incorporate the driver into your kernel source or build it outa tree? And Its been a while since Ive updated the [...]

I downloaded your entire 5.18 kernel source tree and compiled that as I figured it would be less problematic than patching an existing kernel. Yeah that was on a high speed port, and no, same same!

strobe84, 2024-03-01:
Morning! I've tried loading the driver without bind232h parameter but changing the serial of the device and unfortunately it's the same behaviour. I'm going to try and dig out the precise version of the FTDI chip I have(should be the same as yours but just checking!) just incase there's a difference in the MPSSE configuration of it.

> [ 7641.205403] usb 1-1: new high-speed USB device number 8 using ehci-pci
[ 7641.529177] usb 1-1: New USB device found, idVendor=0403, idProduct=6014, bcdDevice= 9.00
[ 7641.529182] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 7641.529184] usb 1-1: Product: ft232H-16ton-i2c
[ 7641.529185] usb 1-1: Manufacturer: FTDI
[ 7641.529186] usb 1-1: SerialNumber: 16ton
[ 7641.538034] i2c-ftdi 1-1:1.0: probe i2cftdi freq 100000
[ 7641.607692] i2c_dev: adapter [FTDI USB-to-I2C at bus 001 device 008] registered as minor 1
[ 7641.607720] i2c i2c-1: adapter [FTDI USB-to-I2C at bus 001 device 008] registered
[ 7641.607727] i2c-ftdi 1-1:1.0: Initialized FTDI-based I2C device
[ 7667.705331] i2c i2c-1: ioctl, cmd=0x705, arg=0x7ffd964173e8
[ 7667.705412] i2c i2c-1: ioctl, cmd=0x703, arg=0x08
[ 7667.705415] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffd96417370
[ 7667.817284] i2c i2c-1: ioctl, cmd=0x703, arg=0x09
[ 7667.817297] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffd96417370
[ 7740.259382] i2c i2c-1: ioctl, cmd=0x705, arg=0x7ffec98c7188
[ 7740.259464] i2c i2c-1: ioctl, cmd=0x703, arg=0x08
[ 7740.259469] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffec98c7110
[ 7751.464889] i2c i2c-1: ioctl, cmd=0x703, arg=0x09
[ 7751.464895] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffec98c7110
[ 7762.570626] i2c i2c-1: ioctl, cmd=0x703, arg=0x0a
[ 7762.570633] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffec98c7110
[ 7775.977547] i2c i2c-1: ioctl, cmd=0x703, arg=0x0b
[ 7775.977555] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffec98c7110
[ 7787.594626] i2c i2c-1: ioctl, cmd=0x703, arg=0x0c
[ 7787.594633] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffec98c7110
[ 7799.629834] i2c i2c-1: ioctl, cmd=0x703, arg=0x0d
[ 7799.629845] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffec98c7110
[ 7810.909541] i2c i2c-1: ioctl, cmd=0x703, arg=0x0e
[ 7810.909548] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffec98c7110
[ 7822.644593] i2c i2c-1: ioctl, cmd=0x703, arg=0x0f
[ 7822.644600] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffec98c7110
[ 7833.992515] i2c i2c-1: ioctl, cmd=0x703, arg=0x10
[ 7833.992522] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffec98c7110
[ 7848.110720] i2c i2c-1: ioctl, cmd=0x703, arg=0x11
[ 7848.110727] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffec98c7110
[ 7858.785737] i2c i2c-1: ioctl, cmd=0x703, arg=0x12
[ 7858.785744] i2c i2c-1: ioctl, cmd=0x720, arg=0x7ffec98c7110

> sudo systool -vm i2c_ftdi
Module = "i2c_ftdi"

  Attributes:
    coresize            = "20480"
    initsize            = "0"
    initstate           = "live"
    refcnt              = "0"
    srcversion          = "7C78FE5AE080A84FE3F2B41"
    taint               = ""
    uevent              = <store method only>

  Parameters:
    100k                = "1"
    1m                  = "0"
    3m                  = "0"
    adaptive            = "0"
    bind232h            = "0"

  Sections:
    .bss                = "0xffffffffc04544c0"
    .data               = "0xffffffffc0454000"
    .exit.text          = "0xffffffffc04523b8"
    .gnu.linkonce.this_module= "0xffffffffc0454140"
    .init.text          = "0xffffffffc0457000"
    .note.Linux         = "0xffffffffc0453024"
    .note.gnu.build-id  = "0xffffffffc0453000"
    .rodata             = "0xffffffffc04532a0"
    .rodata.str1.1      = "0xffffffffc04530e4"
    .rodata.str1.8      = "0xffffffffc0453188"
    .strtab             = "0xffffffffc0458c60"
    .symtab             = "0xffffffffc0458000"
    .text               = "0xffffffffc0451000"
    .text.unlikely      = "0xffffffffc04520b6"
    __mcount_loc        = "0xffffffffc0453054"
    __param             = "0xffffffffc04533b8"

bm16ton, 2024-03-01:
just curious what i2c devices are attached to your ft232h? Also did u try
any speeds besides 100k? Doubtful its a clock issue but maybe worth trying.

strobe84, 2024-03-01:
> just curious what i2c devices are attached to your ft232h? Also did u try any speeds besides 100k? Doubtful its a clock issue but maybe worth trying.
> […](#)

Hey, sorry... yes there's a MAX31875 attached to the bus and a PAC1951T-2E power monitor. Changing the clock to 400k made no difference.

bm16ton, 2024-03-02:
I am totally lost. And with the userland drivers your also able to scan the
i2c bus, but with the userland drivers it acts identical to the kernel
drivers except it shows a hit at whatever address your i2c peripheral
device uses?

bm16ton, 2024-03-03:
Just now realizing how strange it is your dmesg log has the cmd and ioctl.
I dont even have the word ioctl in my driver. Normally when using a pure
userland driver i could enable the kernels built in usbsnoop to get the
ioctl's,  but as soon as id switch to kernel driver all it would show well
nothing. I willingly admit i may have been doing something wrong, and did
it mostly on spi drivers, which compared to an i2c bus scan keep much more
of it actions inside the kernel. Just curious what mechanism you used? Of
course in recent history kernel debuging msgs have gone thru changes,
muddying the waters even more.

strobe84, 2024-03-04:
Morning! By means of an update, I think I'm going to have to park this for now as i've kinda "timed out". When I have some slower time, I'm going to build your module with a clean or newer source tree than 5.18 and see what happens. I have a feeling this is where the issue may lie but not entirely sure. One for the backburner but appreciate your help and has dusted off some knowledge I'd parked some time back!!!

closed 2024-03-04 by strobe84

bm16ton, 2024-03-04:
i understand. Im currently (and usually) running the latest version of
kernels including rc's with this driver, and dont always remember to update
git repo, so just drop me a line and ill make sure you have the latest code.