bm16ton/ft2232-mpsse-i2c-spi-kern-drivers #7: Need help getting SPI to work on FT2232H mini module

megaloegoman, 2024-05-27:
I 1st tried ft232h-app to enable  SPI but got a kernel crash. See dmesg.txt
I also saw this error, so I tried manually inserting the modules, but it didn't make a difference
_cat: /tmp/irqper: No such file or directory
cat: /tmp/usbwait: No such file or directory_

sudo modprobe -r i2c-ftdi
sudo modprobe -r spi_ftdi_mpsse ft232h_intf
sudo modprobe ft232h-intf bind232h=1  poll_period=10 irqpoll=1

Configuring it as I2C does work - I see 2 I2C interfaces created.

[dmesg.txt](https://github.com/bm16ton/ft2232-mpsse-i2c-spi-kern-drivers/files/15450356/dmesg.txt)

I'm using ubuntu 22 and kernel 5.19. Would greatly appreciate any help.

megaloegoman, 2024-07-29:
Figured it out. Only had to change the name of the device.
[change_name.patch](https://github.com/user-attachments/files/16406173/change_name.patch)

Can you have it recognize both names?

bm16ton, 2024-07-29:
yeah i purposely chose to use the non default names so it wouldnt interfere
with the regular use cases, and so people would be forced to know what tgey
where doing before they ised the drivers. super glad you got it working i
love theae things.

megaloegoman, 2024-07-29:
So can you change it to recognize both names? I don't think most users know how to flash the firmware. I've never done it.

I like it too because it's the only solution that has SPI, I2C and GPIO, which the SOC I'm working with needs. Unforuntately, I realized that GPIO interrupts can only be simulated with polling because USB <= 2  doesn't support interrupts.

bm16ton, 2024-07-29:
Its possible, ill have to think on it, currently the names are used to
determine when things like the usb to serial driver for ftdi should ignore
the device. Tho i do have kernel module parameters for the various modules
to bypass said restrictions. I use those in particular for the ft232 since
it only has the single interface, and I wanted switching between
i2c/spi/uart to be easy, on my machine when the 232series is plugged in a
gui comes up and i can select wich protocol to use. Plus im years away from
ever even attempting at getting any driver mainlined in kernel. And i know
they would never accept a driver if it had competing drivers IE ftdi-sio vs
i2c or spi. But yeah im willing to consider it, or maybe even a simple
module param etc. You my friend have giving me much to consider:) I too use
these not only on laptops and desktops but sbc's/handheld's and the
difference between writing userland drivers for every possible device,
which has more points of failure and is slower etc is huge. Had ftdi not
wanted vendor lock-in so badly they would have written kernel drivers. I do
see some newer ftdi usb-2-i2c have mainline driver support. Oh jeez its
late i just realized did the irq not work for you in the end? i have a
small number of spi devices and fewer with irq requirements. Its been a
while since i worked on this project. I believe some devices would want the
gpio pre-setup correctly (input, active hi/low etc) and simply use the irq
it triggers, and others would want to set it up themselves. I thought i had
it worked out for either scenario but i may need to recheck that. Let me
know and if irq isnt working ill get it going for yeah. And your absolutely
correct about most people not knowing how to update the eeprom, i should
have written a gui etc for that ages ago. Currently the open source eeprom
writing app has little to know docs and it confuses people to no end.